risinek/esp32-wifi-penetration-tool #7: Provide guidance on how to use hashcat

risinek, 2021-05-07:
Document step by step how to start password recovery using Hashcat from HCCAPX or PMKID (16800)

risinek, 2021-05-07:
PMKID:
1. save the PMKID in hashcat format to some plain text file (e.g. test.16800)
2. run ./hashcat -m 16800 test.16800 -a 3 (or whatever attack type you want to run)

andrewostroumov, 2021-08-13:
When I try to penetrate my own wifi Xiaomi_G I got `dc3324a111e61044ad3fd14adda9b33f*240ac4f93888*4a8f5a800f32*5869616f6d695f47`

With `
sudo hashcat-6.2.3/hashcat.bin -m 16800 -a3 -w3 --force --increment "dc3324a111e61044ad3fd14adda9b33f*240ac4f93888*4a8f5a800f32*5869616f6d695f47" "samsung7"
`
I got exhausted status

I already try with dictionaries instead of brute force - no results
Move PMKID to file and use this file with hashcat - no results

I think it's something with process of fetch PMKID

```
ESP32 Wi-Fi Penetration Tool

FINISHED
ATTACK_TYPE_PMKID
MAC AP: 240ac4f93888
MAC STA: 4a8f5a800f32
(E)SSID: 5869616f6d695f47 (Xiaomi_G)
PMKID #0: dc3324a111e61044ad3fd14adda9b33f

Hashcat ready format:
PMKID #0: dc3324a111e61044ad3fd14adda9b33f*240ac4f93888*4a8f5a800f32*5869616f6d695f47
```

arisalpharian, 2023-03-13:
> 

try change MAC AP & MAC STA position in hashcat ready format. In your case, its look like:
dc3324a111e61044ad3fd14adda9b33f*4a8f5a800f32*240ac4f93888*5869616f6d695f47

arisalpharian, 2023-03-13:
[image]
Switch MAC AP & MAC STA position

StergiosT0, 2023-09-19:
How can i see the wi-fi key?