risinek/esp32-wifi-penetration-tool #8: CAP file does not contain ESSID and handshake.

heshxxx, 2021-05-27:
After attacking the hccapx file is good. And in the "CAP" file there is no handshake.

risinek, 2021-05-27:
So the HCCAPX worked fine and was crackable, but pcap file was empty? Would you mind sharing both files? 

PCAP does not include ESSID, because ESSID is not part of EAPoL-Key frames (4-way handshake frames), so that's expected. PCAP file should only include EAPoL-Key frames. If you run cap2hccapx script on that pcap file, it will fail and thats expected.

heshxxx, 2021-05-27:
>  Would you mind sharing both files?

[cap.zip](https://github.com/risinek/esp32-wifi-penetration-tool/files/6553197/cap.zip)

risinek, 2021-05-27:
I see 59 packets in the pcap file, so how did you end up with the conclusion that there are no handshake packets? 
Did you try to run `cap2hccapx` script on it or something similar from hashcat tools? It will not work, because pcap only contains EAPoL-Key frames as those are the actual 4-way WPA/WPA2 handshake frames. It's "by design" this way, but it can be a feature request to include also association frames into pcap, so that it will work with cap2hccapx script (but I don't see a reason for this atm as hccapx file is provided directly by this tool).
[image]

closed 2021-05-27 by heshxxx