risinek/esp32-wifi-penetration-tool #9: Add support for MicroSD

RChadwick7, 2021-05-27:
It could be extremely useful to be able to store handshakes and PMKID's on an external MicroSD.

risinek, 2021-05-30:
MicroSD card would require additional HW, but I'm thinking about saving captured handshakes/pmkids into flash memory (which would be optional to prevent flash wear). 
MicroSD support can be done later 👍

RChadwick7, 2021-05-30:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAADAAAAAwCAYAAABXAvmHAAAAAXNSR0IArs4c6QAAAARzQklUCAgICHwIZIgAAAAfSURBVGiB7cEBAQAAAIIg/69uSEABAAAAAAAAAAAvBiQwAAGO68BnAAAAAElFTkSuQmCC###24x24:true###][Tracking image][https://github.com/notifications/beacon/ACA4C2ZXFENUB3CE6VY65IDTQJE33A5CNFSM45U3LEBKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOGK4WJEA.gif]

The additional hardware doesn't need to be difficult to add. MicroSD modules are cheap, and would only require soldering a few wires. If you plan on growing the tool, I think eventually the lack of SD will be a roadblock. If you're not a hardware person, but willing to figure out the software, I could come up with part numbers and directions on how to easily add a MicroSD.