risinek/esp32-wifi-penetration-tool #14: Error "Separator unmatched" with hashcat

QuacNoobLoL, 2021-09-06:
I have error "separator unmatched" when trying to use handshake hccapx with hashcat mode 22000
I generated the file using my own access point.

Output:
`PS F:\hashcat-6.2.4> hashcat -m 22000 capture1.hccapx wordlist.txt
hashcat (v6.2.4) starting

Successfully initialized NVIDIA CUDA library.

* Device #1: CUDA SDK Toolkit not installed or incorrectly installed.
             CUDA SDK Toolkit required for proper device support and utilization.
             Falling back to OpenCL runtime.

* Device #1: WARNING! Kernel exec timeout is not disabled.
             This may cause "CL_OUT_OF_RESOURCES" or related errors.
             To disable the timeout, see: https://hashcat.net/q/timeoutpatch
nvmlDeviceGetFanSpeed(): Not Supported

OpenCL API (OpenCL 3.0 CUDA 11.4.125) - Platform #1 [NVIDIA Corporation]
========================================================================
* Device #1: NVIDIA GeForce GTX 1660 Ti, 5376/6144 MB (1536 MB allocatable), 24MCU

OpenCL API (OpenCL 2.1 ) - Platform #2 [Intel(R) Corporation]
=============================================================
* Device #2: Intel(R) UHD Graphics 630, 1568/3214 MB (803 MB allocatable), 24MCU

Minimum password length supported by kernel: 8
Maximum password length supported by kernel: 63

Hashfile 'capture1.hccapx' on line 1 (HCPX♦): Separator unmatched
Hashfile 'capture1.hccapx' on line 2 ("·↑úæF┬m▒∟#◄∟ å/╧èû∞$): Separator unmatched
Hashfile 'capture1.hccapx' on line 3 (): Separator unmatched
No hashes loaded.

Started: Mon Sep 06 12:42:23 2021
Stopped: Mon Sep 06 12:42:24 2021`
[capture.zip](https://github.com/risinek/esp32-wifi-penetration-tool/files/7113602/capture.zip)

QuacNoobLoL, 2021-09-06:
Hmm i watched the demo video and found out that I'm missing --potfile-disable. Now it is running. But why can't we enable it?

QuacNoobLoL, 2021-09-06:
lmao i used the wrong file and now the problem persists :>

risinek, 2021-09-10:
you are using wrong mode. for hccapx files you should use mode 2500. See "Preparation" section on this wiki https://hashcat.net/wiki/doku.php?id=cracking_wpawpa2#preparation

maalos, 2021-09-11:
though when using 2500 it says it's deprecated and you have to use 22000 instead

maalos, 2021-09-11:
ok i figured it out, when using 2500 add `--deprecated-check-disable`

risinek, 2021-09-11:
Good to know, thanks! Feel free to add it somewhere to the READMEs.
And adding support for this new recommended hash mode will be worthy. I will make this an feature request eventually.

QuacNoobLoL, 2021-10-18:
There's another way for this and still using the mode 22000. I have to extract the info from the hccapx file to a WPA hash string like this in https://hashcat.net/wiki/doku.php?id=cracking_wpawpa2#working_with_hash_files

bharathkumaracharips, 2023-06-07:
> ok i figured it out, when using 2500 add `--deprecated-check-disable`

give complete code bro

hosemigel-ux, 2023-06-17:
try this:  hashcat.exe -m 2500 -a 3 {file name} ?d?d?d?d?d?d?d?d --deprecated-check-disable