risinek/esp32-wifi-penetration-tool #15: Cannot download any file from the site

maalos, 2021-09-11:
as in title, im unable to do so. i have an esp32 (not s2/c3) and im getting that sort of an error when following the demo vid
[image]

maalos, 2021-09-11:
ok it's only happening on a phone, but the thing is i need to use it on a phone

maalos, 2021-09-11:
only happening on mobile chrome, mobile firefox works fine

risinek, 2021-09-11:
Yes I know about this and this is unfortunately limitation of http_client lib. Some browsers are a bit verbose in user agent header usually. I can't remember now what helped me when I ran into the same with my Android phone, but I will try to. 
For me it definitely worked in the end with Android Chrome.
Maybe look for user agent settings in your browser, it may be customisable.

maalos, 2021-09-13:
ok so now it doesn't even work on firefox anymore, idk what and how, I've reset the browser, but haven't reflashed it yet

maalos, 2021-09-13:
forgot to mention, the site is here but the long stream of characters that is visible on my PC and the hccapx file aren't there, when trying to download the hccapx file it just loads indefinetely, and the pcap file seems to not have any captured packets

maalos, 2021-09-13:
tested on different networks, only works on mine

risinek, 2021-09-13:
Ok, this is different issue from the headers and it's probably a bug/missing signaling that the capture is empty.
I will mark this as a bug. But if there are no packets in pcap, then no handshake packets were captured and hence hccapx would not be available anyway.

maalos, 2021-09-13:
what can i do then? might that be because of a firewall or do i have to set the capture packet for longer? or maybe like force the handshake by trying to connect with my phone?

maalos, 2021-09-13:
[image]

maalos, 2021-09-13:
[image]

risinek, 2021-09-13:
What devices are connected to the target network? They may resist to deauthentication attack. What is the target network encryption?

maalos, 2021-09-13:
ok, so I've examined the network further, it's WPA2 iirc, and I'm not sure if there are any wireless devices connected to it, since every computer here is connected by ethernet, though teachers do bring their laptops but in total there are about 8 different networks, i think i know which ones are for who because of their unique names but still unsure which one i should try to attack. i did not try to attack the teacher network yet, but i will try tomorrow. does being connected via ethernet mean i can't capture any packets?

Rffy, 2021-09-14:
It only works if the wifi that you're attack is compatible and hccapx have capture then download will be available...

I think This tools is only works in wifi without Protected management frame supported modems/routers..

risinek, 2021-09-15:
> does being connected via ethernet mean i can't capture any packets

this project is focused on attacks on Wi-Fi, specifically WPA/WPA2 authentication. If there are no clients connected, you can try to gather PMKIDs, but not all routers will send them. To execute classic deauth attack, you need some active clients connected to target network via Wi-Fi interface, not wired one.

>if the wifi that you're attack is compatible with hccapx

hccapx is just a file format by Hashcat tool in which WPA/WPA2 handshake data are stored.

> I think This tools is only works in wifi without Protected management frame

that's correct. if protected management frame feature is established between client and router, then deauth attack won't work, but you still should be able to capture handshakes. That means you can't actively attack on the target network, but you can passively listen to handshakes (of course this would need unlimited/long timeout as you suggested in https://github.com/risinek/esp32-wifi-penetration-tool/issues/5#issuecomment-918720316)

maalos, 2021-09-16:
well, now that's the longest question addressing answer I've ever seen, but back to the topic, thank you for answering all of my questions. I'm probably going to wait for the loop then, or try to continuously attack with the 255s timeout. i know it may not be legal, but as they say, it's in the name of science, and just maybe there will be a vulnerability, though i would be afraid to tell anyone about it xD. i know what the difference between wifi and ethernet is. i just read what pmkid is, and I'm probably going to try that. also, how the hell will they know who is connected if my hostname isn't my name and surname? either way, i know it sounds really unethical and all to attack a private teacher network, and i can promise you (ik you don't care) that there will be no malicious misuse of it, just a form of practice.

closed 2021-09-16 by maalos