risinek/esp32-wifi-penetration-tool #16: does not turn on after loading the code

ddabra, 2021-09-16:
I uploaded 3 binaries based on the instructions into my nodeMCU, but after reboot, the AP does not appear.

Mikewins1, 2021-10-04:
Im having the same problem.  I used the command you had in the instructions and after everything was done I never see an AP to connect to, not even a hidden network.

risinek, 2021-10-05:
Did you check logs, was there something suspicious, aren't there any critical failures causing constat reboots? Can you share logs maybe? What power source are you using?

YadneshD, 2021-12-29:
Hi, Can we install it in nodemcu esp8266 version 1.0 using same steps for esp32? Are any changes required while installing?

YadneshD, 2021-12-29:
> Did you check logs and was there something suspicious, aren't there any critical failures causing constat reboots? Can you share logs maybe? What power source are you using?

I tried it on my nodeMCU ESP8266. There is no new WiFi network named management AP. I can't find any new network on my mobile, PC wiFi. The bin files are properly uploaded on the SoC. Power supply is given by normal USB port of my PC. I cant find anything on serial monitor of arduino. I have attached a screenshot for your reference of powershell window through which I uploaded the bin files. I didn't build your repo, instead I uploaded the bin files using esptool python package. Please tell me if you need more details.

[image]

davvi, 2022-04-21:
Same problem. I guess it about NVS

Here is the full log:

```

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7152
load:0x40078000,len:14292
ho 0 tail 12 room 4
load:0x40080400,len:3688
entry 0x40080678
I (29) boot: ESP-IDF 4.3.0 2nd stage bootloader
I (29) boot: compile time 13:21:10
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1a9b0h (108976) map
I (152) esp_image: segment 1: paddr=0002a9d8 vaddr=3ffb0000 size=05050h ( 20560) load
I (161) esp_image: segment 2: paddr=0002fa30 vaddr=40080000 size=005e8h (  1512) load
I (162) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=7c39ch (508828) map
I (360) esp_image: segment 4: paddr=000ac3c4 vaddr=400805e8 size=14688h ( 83592) load
I (396) esp_image: segment 5: paddr=000c0a54 vaddr=50000000 size=00010h (    16) load
I (408) boot: Loaded app from partition at offset 0x10000
I (408) boot: Disabling RNG early entropy source...
I (420) cpu_start: Pro cpu up.
I (420) cpu_start: Starting app cpu, entry point is 0x400811f0
I (406) cpu_start: App cpu up.
I (434) cpu_start: Pro cpu start user code
I (435) cpu_start: cpu freq: 160000000
I (435) cpu_start: Application information:
I (439) cpu_start: Project name:     HACKWF
I (444) cpu_start: App version:      1
I (448) cpu_start: Compile time:     Apr 21 2022 13:20:43
I (454) cpu_start: ELF file SHA256:  c8e4e100e6bc8e32...
I (460) cpu_start: ESP-IDF:          4.3.0
I (465) heap_init: Initializing. RAM available for dynamic allocation:
I (472) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (478) heap_init: At 3FFB9048 len 00026FB8 (155 KiB): DRAM
I (485) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (491) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (497) heap_init: At 40094C70 len 0000B390 (44 KiB): IRAM
I (505) spi_flash: detected chip: gd
I (508) spi_flash: flash io: dio
I (513) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
D (523) main: app_main started
D (523) wifi_controller: Starting AP...
I (533) wifi_init: rx ba win: 6
I (533) wifi_init: tcpip mbox: 32
I (533) wifi_init: udp mbox: 6
I (543) wifi_init: tcp mbox: 6
I (543) wifi_init: tcp tx win: 5744
I (553) wifi_init: tcp rx win: 5744
I (553) wifi_init: tcp mss: 1440
I (553) wifi_init: WiFi IRAM OP enabled
I (563) wifi_init: WiFi RX IRAM OP enabled
ESP_ERROR_CHECK failed: esp_err_t 0x1101 (ESP_ERR_NVS_NOT_INITIALIZED) at 0x40086980
file: "src/wifi_controller/wifi_controller.c" line 39
func: wifi_init_apsta
expression: esp_wifi_init(&wifi_init_config)

abort() was called at PC 0x40086983 on core 0

Backtrace:0x400878d7:0x3ffbba70 0x400880a9:0x3ffbba90 0x4008ee9e:0x3ffbbab0 0x40086983:0x3ffbbb20 0x400d4b36:0x3ffbbb40 0x400d4bdb:0x3ffbbc20 0x400d4c82:0x3ffbbc40 0x400d449e:0x3ffbbcf0 0x4014c0b9:0x3ffbbd10 0x4008b561:0x3ffbbd30

ELF file SHA256: c8e4e100e6bc8e32

Rebooting...
```

davvi, 2022-04-21:
`#include "nvs_flash.h" // just add this line 

/**
 * @brief Initializes Wi-Fi interface into APSTA mode and starts it.
 * 
 * @attention This function should be called only once.
 */
static void wifi_init_apsta(){
    ESP_ERROR_CHECK(nvs_flash_init()); // and this line 

`

hardcod3dd, 2022-04-21:
Im having the same issue.How can i take logs like yours ?

> Same problem. I guess it about NVS
>
> Here is the full log:
>
> ```
>
> ets Jun  8 2016 00:22:57
>
> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee [...]

Im having the same issue.How can i take logs like yours ?

hardcod3dd, 2022-04-22:
../components/wifi_controller/wifi_controller.c:13:10: fatal error: nvs_flash.h: No such file or directory
 #include "nvs_flash.h"

risinek, 2022-05-04:
NVS is disabled in sdkconfig file https://github.com/risinek/esp32-wifi-penetration-tool/blob/1874eca01d98eb6466de00077aa417d980878d75/sdkconfig.defaults

azhar25785, 2023-07-31:
i had tgat same issue after uploading i pressed the boot and en buttons on esp32 and restart your laptop wifi it should show then

xmodpt, 2023-10-01:
Hello every one

Great job with the app however i am having problems with it.
i have just clone the repo, followed the video from youtube (   ) and all when fine with out any errors, but when trying to connect to the AP, there isn't any... the AP is not found.

Also i have tryed to set sdkconfig.defaults file from "CONFIG_ESP32_WIFI_NVS_ENABLED=n" to "y" and uploaded it again and still nothing.

just in case i uploaded the blinking sample from arduino IDE and my nodeMCU 8266 works perfectly.

can any one give me a hint on how to fix this ?

regards to all

NP

risinek, 2023-10-01:
This is ESP32 project, not 8266. How did you even managed to flash this project to 8266?

xmodpt, 2023-10-01:
Sorry my bad. I mennt esp32.
I mixed up the text about node mcu in another forum

risinek, 2023-10-01:
without any logs I cannot help. Can you provide logs after flashing and program logs from runtime?