risinek/esp32-wifi-penetration-tool #20: changing AP password

youhacker55, 2021-11-19:
can you add options to control ap  security(wpa,wep) , change AP password

rclough, 2022-01-31:
You can change it in the config in your local files before compiling and uploading to your ESP32. It's located in `components/wifi_controller/Kconfig`: https://github.com/risinek/esp32-wifi-penetration-tool/blob/1874eca01d98eb6466de00077aa417d980878d75/components/wifi_controller/Kconfig#L9-L19

Lan029001, 2022-07-09:
dont work

francoil, 2023-04-24:
didn't work

azhar25785, 2023-07-31:
didn't work ....tried many times i think code must be built after changes but how to rebuild it

risinek, 2023-07-31:
Yes, project has to be rebuild. How to do that is in README.

azhar25785, 2023-07-31:
Got this error while compiling **esp_event.h library not found**

C:/Users/azhar/Desktop/32deoot/esp32-wifi-penetration-tool-master/components/frame_analyzer/interface/frame_analyzer.h:12:10: fatal error: esp_event.h: No such file or directory
   12 | #include "esp_event.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
[887/904] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/attack.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\azhar\Desktop\32deoot\esp32-wifi-penetration-tool-master\build\log\idf_py_stderr_output_12300 and C:\Users\azhar\Desktop\32deoot\esp32-wifi-penetration-tool-master\build\log\idf_py_stdout_output_12300
PS C:\Users\azhar\Desktop\32deoot\esp32-wifi-penetration-tool-master>

mualiyu, 2023-08-04:
yes, I am having the same Issue too

albinvar, 2024-05-18:
The configuration is stored in the `sdkconfig` file located at the root of the project, which should be generated after configuring your project using `idf.py menuconfig` ([#refrence](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#project-configuration-menu))

To configure the Wi-Fi AP SSID and password for your ESP32 Jammer, follow these steps:

- Perform a full clean for the builds using `idf.py fullclean` command.
- To change the default ap ssid/password, run `idf.py menuconfig`
- select `component config --->`  and select `Wi-Fi Controller  --->`, then `Management AP`
- You can now customize your ssid and password from here.

[image]

- After customizing your config, hit `S` to save the configuration file.
- [Rebuild](https://github.com/risinek/esp32-wifi-penetration-tool?tab=readme-ov-file#build) your project and [flash](https://github.com/risinek/esp32-wifi-penetration-tool?tab=readme-ov-file#Flash) it to your board