risinek/esp32-wifi-penetration-tool #25: fails to build - Error 2

1stmemberoftheDKcrew, 2022-03-11:
```
[ 93%] Building C object esp-idf/hccapx_serializer/CMakeFiles/__idf_hccapx_serializer.dir/hccapx_serializer.c.obj
/Users/XXXX/Downloads/esp32-wifi-penetration-tool-master/components/hccapx_serializer/hccapx_serializer.c: In function 'save_eapol':
/Users/XXXX/Downloads/esp32-wifi-penetration-tool-master/components/hccapx_serializer/hccapx_serializer.c:102:49: error: implicit declaration of function 'ntoh' [-Werror=implicit-function-declaration]
     eapol_len = sizeof(eapol_packet_header_t) + ntohs(eapol_packet->header.packet_body_length);
                                                 ^~~~~
cc1: some warnings being treated as errors
make[3]: *** [esp-idf/hccapx_serializer/CMakeFiles/__idf_hccapx_serializer.dir/hccapx_serializer.c.obj] Error 1
make[2]: *** [esp-idf/hccapx_serializer/CMakeFiles/__idf_hccapx_serializer.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
```
then a couple lines later...
```
make[1]: *** [CMakeFiles/flash.dir/rule] Error 2
make: *** [flash] Error 2
make failed with exit code 2
```
I'm on mac os 10.14.6

smpplaza, 2022-03-15:
I have only Error 1 but it's practically the same issue: 

`[2/11] Building C object esp-idf/hccapx_serializer/CMakeFiles/__idf_hccapx_serializer.dir/hccapx_serializer.c.obj
FAILED: esp-idf/hccapx_serializer/CMakeFiles/__idf_hccapx_serializer.dir/hccapx_serializer.c.obj 
`

And later: 

`     eapol_len = sizeof(eapol_packet_header_t) + ntohs(eapol_packet->header.packet_body_length);
                                                 ^~~~~
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
`

smpplaza, 2022-05-12:
BTW this is on the Wemos S2 mini. Somehow this is not working on that board...

smpplaza, 2022-05-12:
I have tried many many ways both on Linux with IDF and on windows with a couple of programmers. I've erased the mem in every try but it will NOT work on the Wemos S2 board...

smpplaza, 2022-05-12:
O.k. so I finally got this working:
- install VSCode on Linux
- install ESP-IDF plugin and leave everything default
- clone the github in VScode
- build in VScode
- make sure every setting in the lower blue bar is set to the correct setting: 
- Select port (on my sys it was /dev/ttyACM0), Select board: ESP32s2, set flash type to UART then click flash...

jwf, 2022-05-24:
VSCode installs a released version (4.4.1 at the moment) instead of checking out master. So likely that's where the difference, and thus this issue lies. I had the same problem, also solved by just pulling the v4.4.1 branch of esp-idf.