risinek/esp32-wifi-penetration-tool #26: working with wemos mini s2?

yentzee, 2022-03-17:
Hi, I am trying to flash a wemos s2 mini using the esp download tool
1. bootloader 0x1000
2. partitiontable 0x8000
3. esp wifi penetration tool 0x10000
40 MHz
DIO
1152000 
and in my case com 10 (?)
just got a esp32 first time today. Not sure what is wrong.
It says sync but nothing happens. I tried resetting with 0 hold, reset hold , release 0
Help is greatly appreciated

risinek, 2022-05-04:
This is not related to this project, you need to find help for general ESP32 flashing

closed 2022-05-04 by risinek

smpplaza, 2022-05-12:
Nono this is not realted to windows only
I've tried this a hundred times on windows as well as on linux... this is absolutly not working on a wemos s2 mini

smpplaza, 2022-05-13:
O.k. so I finally got this working:
- install VSCode on Linux
- install ESP-IDF plugin and leave everything default
- clone the github in VScode
- build in VScode
- make sure every setting in the lower blue bar is set to the correct setting: 
- Select port (on my sys it was /dev/ttyACM0), Select board: ESP32s2, set flash type to UART then click flash...