risinek/esp32-wifi-penetration-tool #37: The plugin 2500 is deprecated and was replaced with plugin 22000.

toto99303, 2022-11-05:
Hello,

Please add support for the new format 22000 as hashcat returns this error and never tries to start cracking:

"The plugin 2500 is deprecated and was replaced with plugin 22000."

Otherwise - Awesome tool, thanks!!

toto99303, 2022-11-05:
Just figured that adding `--deprecated-check-disable` ignores the error, but it would be nice if you add the new format. Thanks!

risinek, 2022-11-09:
Sounds like a feasible feature request. Have you come across a new format specification by any chance?

toto99303, 2022-11-09:
Yes, here:
https://hashcat.net/forum/thread-10253.html

For developers

The new hash format 22000 in detail:

Code:
PROTOCOL*TYPE*PMKID/MIC*MACAP*MACCLIENT*ESSID*ANONCE*EAPOL*MESSAGEPAIR

PROTOCOL = Fixed string "WPA"
TYPE = 01 for PMKID, 02 for EAPOL
PMKID/MIC = PMKID if TYPE=01, MIC if TYPE=02
MACAP = MAC of AP
MACCLIENT = MAC of CLIENT
ESSID = network name (ESSID) in HEX
ANONCE = ANONCE
EAPOL = EAPOL (SNONCE is in here)
MESSAGEPAIR = Bitmask:
0: MP info (https://hashcat.net/wiki/doku.php?id=hccapx)
1: MP info (https://hashcat.net/wiki/doku.php?id=hccapx)
2: MP info (https://hashcat.net/wiki/doku.php?id=hccapx)
3: x (unused)
4: ap-less attack (set to 1) - no nonce-error-corrections necessary
5: LE router detected (set to 1) - nonce-error-corrections only for LE necessary
6: BE router detected (set to 1) - nonce-error-corrections only for BE necessary
7: not replaycount checked (set to 1) - replaycount not checked, nonce-error-corrections definitely necessary

toto99303, 2022-11-09:
Also here:
https://hashcat.net/wiki/doku.php?id=cracking_wpawpa2

risinek, 2022-11-09:
Thanks! 👍 I can't promise when I will find some time for this, but maybe someone might do it as well. It should be easy as IIRC all the values are already in the binary format, so it's just about formatting them on UI.

Keyvantajfar, 2023-11-21:
> Yes, here: https://hashcat.net/forum/thread-10253.html
>
> For developers
>
> The new hash format 22000 in detail:
>
> Code: `PROTOCOL*TYPE*PMKID/MIC*MACAP*MACCLIENT*ESSID*ANONCE*EAPOL*MESSAGEPAIR`
>
> PROTOCOL [...]

Hi, i am a newbie in this industry, as a hobby, i'd love to add this feature to this board, i just was able to attack my own home network, which i am authorized to because i own the devices. i'd love to add the -m 22000 (this file type) on this hardware so we can also download these new files. if you or anyone could help me in this path, i'd love to contribute to this project, to address this new feature/issue . thanks for your time.