risinek/esp32-wifi-penetration-tool #43: adding new attacks

anmayraj, 2023-01-06:
https://github.com/EParisot/ESP32_Network_Toolbox_App       i found here an app which can literally do many things if we will connect our esp32 via usb or otg but the thing is it works only on the esp32 box this guy made can you add some codes or something so that normal esp32 can also connect to it because this app detects the esp32 but doesn't get connected

risinek, 2023-01-06:
No I cannot provide support for someone's else project. If you cannot figure it out by yourself, you should learn more about ESP development first.

closed 2023-01-06 by risinek

anmayraj, 2023-01-06:
am not going to share it in either way i just want to test that app but unfortunately it doesn't work that's why i am asking if you can help me with this