risinek/esp32-wifi-penetration-tool #44: the code you provided i am facing issue

anmayraj, 2023-01-06:
the only attack i am able to perform is attack_type_handshake and attack method is deauth_rogue_ap rest all does literally nothing then one lat dos attack which combine all option again disconnect my device from router when started why is it happening and how to fix this?

risinek, 2023-01-06:
It depends on your test environment, mostly on what protocols is the target Wi-Fi is using for securing the packets. 

> disconnect my device from router when started
you mean your device from the target Wi-Fi, or your device from ESP32 management hotspot?

anmayraj, 2023-01-06:
i mean to say what it does is performing a deauthentication attack either a choose rogue ap or dos combined all or any option in that segment.

anmayraj, 2023-01-06:
i have a confusion also though it will be nice if you can just tell me a bit that what each attack do because without ino file i can't make much modifications

risinek, 2023-01-09:
I can't provide much guidance on this, but here are some hints where to start
- brief attack implementation description is in the README here https://github.com/risinek/esp32-wifi-penetration-tool/tree/master/main
- brief theory about the attacks https://github.com/risinek/esp32-wifi-penetration-tool/blob/master/doc/ATTACKS_THEORY.md
- go through header files and read comments (or generate doxygen API docs if you prefer)
- learn more about ESP-IDF to understand the code https://docs.espressif.com/projects/esp-idf/en/latest/esp32/
- paper about this tool https://excel.fit.vutbr.cz/submissions/2021/048/48.pdf

closed 2023-01-09 by risinek