risinek/esp32-wifi-penetration-tool #53: Attacks do not appear to be working, ESP32 is not collecting any packet data.

LucidEye, 2023-03-15:
No attacks appear to be working when testing against my home router.  I let attacks run for up to 5 minutes and when the attack timer is over there is no packet data collected.  No data appears on the screen. The PCAP file will download, but it is always empty and the same size (24 bytes).  The HCCAPX file will not download, any attempt to download it results in a download that never downloads anything and never times out.  When trying an active de-auth attack, nothing gets kicked off my router like it should, so nothing here seems to be working at all.  Can someone please help me figure out if I am doing something wrong, or is this project just defunct?  Thank you.

LucidEye, 2023-03-15:
I guess the "problem" is that my routers all use protected management frames and are no longer susceptible to this kind of deauth attack?  How can we make the ESP32 deauth attack clients instead of the router?

risinek, 2023-03-15:
If your network uses protected management frames (802.11w), then the deauth frames are encrypted. This makes it impossible to create and inject valid deauthentication frames to disconnected client (STA) as the client would ignore any unprotected deauthentication frames.

In deauthentication attack you not attacking router, you are basically acting as a router. But as the client (STA) and router (AP) are already communicating together, they have had agreed on some requirements. i.e. STA knows that it should accept only encrypted frames if they agreed on 802.11w usage.

closed 2023-03-15 by risinek