risinek/esp32-wifi-penetration-tool #54: ninja failed with exit code 1 while trying to build

ZeroTwoWeeb, 2023-03-16:
when building I get:
```
/home/${USER}/esp/esp32-wifi-penetration-tool/components/wifi_controller/interface/../ap_scanner.h:12:10: fatal error: esp_wifi_types.h: No such file or directory
   12 | #include "esp_wifi_types.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```
I tried building on windows as well as Linux

ZhengLinLei, 2023-05-07:
I having this issue too.

risinek, 2023-05-07:
do you have ESP IDF correctly installed? Are you able to build some example projects from ESP-IDF?

ZhengLinLei, 2023-05-07:
I cannot use `idf.py` v4.1 because I'm using MacOS but I solve it changing some code, updating the CMakeList for new version of `idf.py` (v5.x)

(And I'm using ESPS2, so the pre-built bin images cannot be used for me)

devwander, 2023-05-09:
I had the same problem, in my case it was solved when I installed an older version of esp-idf (I opted for 4.2).

My system:
Edition: Windows 10 Home Single Language
Version: 21H2
SO: 19044.2846

risinek, 2023-05-10:
Well this project was build with 4.1 and newer versions were not tested, nor are "supported". If you manage to make it work with newer version then 4.1 feel free to update README 👍

mach9977, 2024-04-21:
I slightly edited cmakelists.txt in components to work with esp-idf 5.1
https://drive.google.com/file/d/15rp9V47SAsMJdZ_icQWxyvLZehNN6km6/view?usp=sharing

Or you can download older esp-idf 4.1
https://web.archive.org/web/20230308085757if_/https://dl.espressif.com/dl/idf-installer/esp-idf-tools-setup-offline-4.1.3.exe

Good Luck :)

ZhengLinLei, 2024-04-21:
> I slightly edited cmakelists.txt in components to work with esp-idf 5.1 https://drive.google.com/file/d/15rp9V47SAsMJdZ_icQWxyvLZehNN6km6/view?usp=sharing
> 

Why don't you fork and merge request your changes?