risinek/esp32-wifi-penetration-tool #60: Tool not working with ESP32-S2-Mini-1

CaptSwirly, 2023-05-26:
Hello, 
When I try to flash the bin files to the ESP32, I am left with an error on the device saying, “Invalid chip id. Expected 2 read 0. Bootloader for wrong chip?” Does it work with an esp32 S2-mini-1 or do I need different files. It would help if there was some way to compile it for Arduino IDE and flash it from there. I am using ESP IDF version 5.0.

risinek, 2023-07-29:
See the README pls. This was developed on  ESP-IDF 4.1. Try applying changes from opened PRs https://github.com/risinek/esp32-wifi-penetration-tool/pull/61 and https://github.com/risinek/esp32-wifi-penetration-tool/pull/58 to see if it helps with your issue. 

As per compiling with Arduino IDE. I'm not aware that it would be possible.

alin180493, 2023-12-05:
Any success with this issue? If yes, can you please share the bin files?