risinek/esp32-wifi-penetration-tool #64: i get error while the code is uploading

azhar25785, 2023-07-20:
even though when i upload cod eto my esp32 from arduino ide it works here is the error in the brackets 
(Uploading stub...
Running stub...
Stub running...
Changing baud rate to 115200
Changed.
Unable to perform XMC flash chip startup sequence (Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)).
FLASH_CRYPT_CNT 0
ABS_DONE_0 False
Compressed 24016 bytes to 14877...
Compressed 3072 bytes to 103...
Compressed 723248 bytes to 447613...

 is stub and send flash finish)

risinek, 2023-07-29:
I don't understand the issue. You said that uploading the code using Arduino IDE works, but the title and error says that there was an error while uploading/flashing the code onto the ESP32. If you have problem with flashing the code, it's not related to this project and you should ask on ESP32 forum for help.

azhar25785, 2023-07-31:
Thanks.... issue resolved my data cable was faulty i think

closed 2023-07-31 by azhar25785