risinek/esp32-wifi-penetration-tool #70: Issues with booting

Jan64X, 2023-09-21:
I'm not 100% sure what esp32 board I have but when I flash it and boot it, it outputs this in PuTTY
[image]
Could somebody help please? I need this for a school project...

risinek, 2023-09-22:
Are you sure you have the ESP IDF setup correctly? The error log shows the path to start with "/home/risinek/esp/..." but I doubt you have same account name as I have. Are you able to build and flash some of the example projects in ESP-IDF?

Jan64X, 2023-09-22:
Well I used the example binaries in releases and used this flash tool: https://shorturl.at/ejNRV . How should I build and flash the projects?

Jan64X, 2023-09-22:
I flashed it again using the binaries in the build/ folder and the Flash download tool, but it still gives me the error in the pic.

Jan64X, 2023-09-22:
Ok, I'll try reinstalling ESP-IDF, I'll get back to you

risinek, 2023-09-22:
After a brief googling, it seems like you have unsupported ESP32 chip? Can you figure out what exactly it is?

Jan64X, 2023-09-22:
Well, I didn't buy an ESP32 direcly. I bought this thing called fiberpunk for my 3D printer and because I didn't need it anymore I looked it up and on their github page it says that its based on ESP32. Heres the repo: https://github.com/fiberpunk1/Beam-ESP32

risinek, 2023-09-25:
does it even have wifi module? Anyway this is not a problem of this project, but rather of the compatibility of your chip with ESP-IDF. If you are not able to flash example projects from ESP-IDF, I can't help. Ask on Espressif forums to get help with setting this up.

closed 2023-09-25 by risinek