risinek/esp32-wifi-penetration-tool #80: The code uploads succesfully but the AP doesnot show up

wxasaebw4evevc, 2023-12-08:
The code uploads succesfully but the AP doesnot show up, i dont know what to do

SP00OKy, 2023-12-22:
```
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
NO XMC flash  detected!
FLASH_CRYPT_CNT 0
ABS_DONE_0 False
Compressed 24016 bytes to 14877...
Compressed 3072 bytes to 103...
Compressed 723248 bytes to 447613...

 is stub and send flash finish
```

Same

nekef, 2024-01-04:
Same here but I don’t know if this works on ESP32-C3. 
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
NO XMC flash  detected!
FLASH_CRYPT_CNT 0
ABS_DONE_0 False
Compressed 24016 bytes to 14877...
Compressed 3072 bytes to 103...
Compressed 723248 bytes to 447613...

 is stub and send flash finish