risinek/esp32-wifi-penetration-tool #83: The pcap handshake does not contain a ESSID

Electro-Gamma, 2024-01-09:
pi@raspberrypi:~/Desktop $ aircrack-ng -w word_list.txt capture.pcap
Reading packets, please wait...
Opening capture.pcap
Read 70 packets.

   #  BSSID              ESSID                     Encryption

   1  1C:BF:CE:7A:1B:E4                            WPA (1 handshake)

Choosing first network as target.

Reading packets, please wait...
Opening capture.pcap
Read 70 packets.

1 potential targets

An ESSID is required. Try option -e.

Quitting aircrack-ng...

Diddle1, 2024-02-20:
Yeah mines saying the same thing...

AHMEDDSALIM, 2024-06-09:
same thing for me

Alexxdal, 2024-11-24:
To create a correct pcap file for aircrack you Need to include al least one Beacon frame that contain the ssid needed for password cracking i dont know of the dev included it, meanwhile you can specify It manually with the option -e "target ssid"

Electro-Gamma, 2025-05-22:
> To create a correct pcap file for aircrack you Need to include al least one Beacon frame that contain the ssid needed for password cracking i dont know of the dev included it, meanwhile you can specify It manually with the option -e "target ssid"

Yes ,very good idea i will fixe with create new repository with that **https://github.com/Electro-Gamma/esp32-handshake-capture**
That is my own WIFI:

```

$ ./aircrack-ng handshake_2.pcap
Reading packets, please wait...
Opening handshake_2.pcap
Read 5 packets.

   #  BSSID              ESSID                     Encryption

   1  XX:BF:CE:XX:1B:XX  inwi Home 4G CC4349       WPA (1 handshake)

Choosing first network as target.

Reading packets, please wait...
Opening handshake_2.pcap
Read 5 packets.

1 potential targets

Please specify a dictionary (option -w).

```

closed 2025-05-22 by Electro-Gamma