risinek/esp32-wifi-penetration-tool #93: error on build wih idf.py

daklon1, 2024-04-18:
idf.py build output - 
Compilation failed because wsl_bypasser.h (in "wsl_bypasser" component) includes esp_wifi_types.h, provided by esp_wifi component(s).
However, esp_wifi component(s) is not in the requirements list of "wsl_bypasser".
To fix this, add esp_wifi to REQUIRES list of idf_component_register call in /home/user/Desktop/esp32-wifi-penetration-tool/components/wsl_bypasser/CMakeLists.txt.
please help

zanderp, 2024-07-12:
```
idf_component_register(
    SRCS "wsl_bypasser.c"
    INCLUDE_DIRS "interface"
    REQUIRES "esp_event" "esp_wifi"
)
target_link_libraries(${COMPONENT_LIB} -Wl,-zmuldefs)
```

zanderp, 2024-07-12:
https://github.com/risinek/esp32-wifi-penetration-tool/pull/61/files

buraev, 2025-06-12:
> https://github.com/risinek/esp32-wifi-penetration-tool/pull/61/files

saving my mind