risinek/esp32-wifi-penetration-tool #100: web interface

AlgoVortex, 2024-06-03:
I can't change the web interface after editing the file(index.html), it just stays the same.

please help.

leorighi, 2024-06-06:
Can you give more details? With prints of what was done?

AlgoVortex, 2024-06-06:
Sure, there is css code that I changed, but maybe I'm doing something wrong with the flashing, because I also tried to change just the header, but nothing changed.

body {
    font-family: 'Arial', sans-serif;
    margin: 20px;
    background-color: #f8f9fa;
    color: #333;
}

h1, h2 {
    color: #007bff;
}

table, th, td {
    border: 1px solid #dee2e6;
    border-collapse: collapse;
    text-align: center;
    background-color: #fff;
}

th, td {
    padding: 15px;
}

tr:hover, tr.selected {
    background-color: #e9ecef;
    cursor: pointer;
}

button {
    background-color: #007bff;
    color: #fff;
    border: none;
    padding: 10px 20px;
    cursor: pointer;
    border-radius: 5px;
    font-size: 16px;
}

button:hover {
    background-color: #0056b3;
}

fieldset {
    border: 1px solid #007bff;
    padding: 10px;
    margin-bottom: 20px;
}

legend {
    color: #007bff;
    padding: 0 10px;
    font-weight: bold;
}

label {
    display: inline-block;
    width: 150px;
    text-align: right;
    margin-right: 10px;
}

select {
    padding: 5px;
    border-radius: 5px;
    border: 1px solid #ced4da;
}

closed 2024-06-09 by AlgoVortex