risinek/esp32-wifi-penetration-tool #101: Header fields are too long for server to interpet

China-Wei, 2024-06-11:
用esp32刷入固件以后进行握手包的抓取，进入管理界面，开启握手包监听（自己的WIFI），然后重连自己的WIF，重新回到管理界面，没有看到握手包信息，点击文件格式进行下载，看到如下提示Header fields are too long for server to interpet，请教我该怎么处理可以进行正常抓包且进行握手包的下载

After flashing the firmware with esp32, grab the handshake packet, enter the management interface, turn on handshake packet monitoring (my own WIFI), then reconnect to my own WIFI, return to the management interface, no handshake packet information is seen, click on file format When downloading, I see the following prompt: Header fields are too long for server to interpet. Please tell me how to handle it. I can capture the packet normally and download the handshake package.

Magauser, 2024-06-20:
i have the same issue, did you find solution?

China-Wei, 2024-06-20:
no. expect a master

asaitov, 2024-07-26:
You may try to increase the maximum size of inbound HTTP headers by appending
```
CONFIG_HTTPD_MAX_REQ_HDR_LEN=1024
``` 
to `sdkconfig.defaults` file and rebuild.

China-Wei, 2024-11-10:
Thanks，I will have a try