risinek/esp32-wifi-penetration-tool #111: no hash

YstalX, 2024-09-13:
Warning: wrong timestamps!
This dump file contains frames with wrong timestamps.
It prevent calculation of EAPOL TIMEOUT values.
That is a bug of the capturing/cleaning tool.

Information: limited dump file format detected!
This file format is a very basic format to save captured network data.
It is recommended to use PCAP Next Generation dump file format (or pcapng for short) instead.
The PCAP Next Generation dump file format is an attempt to overcome the limitations
of the currently widely used (but very limited) libpcap (cap, pcap) format.
https://www.wireshark.org/docs/wsug_html_chunked/AppFiles.html#ChAppFilesCaptureFilesSection
https://github.com/pcapng/pcapng

Information: radiotap header is missing!
Radiotap is a de facto standard for 802.11 frame injection and
reception. The radiotap header format is a mechanism to supply
additional information about frames, rom the driver to userspace
applications.
https://www.radiotap.org/

Information: missing frames!
This dump file does not contain BEACON or PROBERESPONSE frames.
This frames contain the ESSID which is mandatory to calculate a PMK.
It always happens if the capture file was cleaned or
it could happen if filter options are used during capturing.
That makes it impossible to recover the PSK.

Information: missing frames!
This dump file does not contain undirected proberequest frames.
An undirected proberequest may contain information about the PSK.
It always happens if the capture file was cleaned or
it could happen if filter options are used during capturing.
That makes it hard to recover the PSK.

Information: missing frames!
This dump file does not contain important frames like
authentication, association or reassociation.
It always happens if the capture file was cleaned or
it could happen if filter options are used during capturing.
That makes it hard to recover the PSK.

Information: no hashes written to hash files

Sad01122, 2024-11-13:
It also happend to me. Unable to crack the password from **hcappx file**. Also 

> when converting pcap to hccapx through hashcat.net's tool, they say info is missing. it can be because of filtering.

 I think it is no longer supported now. The code needs some changes.

alejandro-atala, 2025-01-21:
Hello, can you explain how to decode the capture.hccapx file? When I want to decipher it with hascat I get the following error 

C:\hashcat>.\hashcat.exe ..\\capture.hccapx --potfile-disable -m 22000 -a 3 123?d?d?d?d?d

Hashfile '.. capture.hccapx' on line 1 (HCPX): Token length exception
Hashfile '.. capture.hccapx' on line 2 (îOäö ̄Ë■ ́=Ì÷jl³þ▓ÈO»âÉ4│FLçíqí▄╦Ôày): Separator unmatched
Hashfile '.. capture.hccapx' on line 3 (): Separator unmatched

* Token length exception: 1/1 hashes
  This error happens if the wrong hash type is specified, if the hashes are
  malformed, or if input is otherwise not as expected (for example, if the
  --username option is used but no username is present)

No hashes loaded.