risinek/esp32-wifi-penetration-tool #118: ESP WROOM 32 Flashing but no wifi is showing

Akshar062, 2024-09-30:
`C:\Users\Krenil\Downloads\bootloader.bin
test offset :  65536 0x10000
case ok
test offset :  65536 0x10000
case ok
test offset :  65536 0x10000
case ok
test offset :  65536 0x10000
case ok
test offset :  65536 0x10000
case ok
test offset :  65536 0x10000
case ok
.
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 115200
Changed.
NO XMC flash  detected!
FLASH_CRYPT_CNT 0
ABS_DONE_0 False
Compressed 24016 bytes to 14877...

 is stub and send flash finish`
 
 
 
 is There issue

dolodi, 2024-10-01:
You have to flash first at 0x1000 the bootloader then partition table at 0x8000 and at 0x10000 the project bin

warlordhariram2008-ops, 2025-09-15:
used those no wifi appearing

Droid-MAX, 2025-11-25:
https://github.com/Droid-MAX/esp32-wifi-penetration-tool/releases/tag/v1.0-dev