risinek/esp32-wifi-penetration-tool #129: Deauth attack duration

Maksotm, 2025-02-02:
Can I set the deauth attack duration to more than 255s? Because there is something limiting that and im wondering if i could remove the limit and if it would affect the esp32's temperature or burn it

Maksotm, 2025-02-02:
shall i change the max value here? 
                    <input type="number" min="0" max="255" id="attack_timeout" value="" required/>

Its in the webserver part, Im wondering if it would work if i changed the max value to 3600 for example

unknown1soldier, 2025-02-22:
> shall i change the max value here? 
>                     <input type="number" min="0" max="255" id="attack_timeout" value="" required/>
> 
> Its in the webserver part, Im wondering if it would work if i changed the max value to 3600 for example

I think it will make the esp32's wifi core overheating and it can be burn (I think I'm not professional)

blaze-cybr, 2025-03-18:
> > shall i change the max value here?
> > 
> > Its in the webserver part, Im wondering if it would work if i changed the max value to 3600 for example
> 
> I think it will make the esp32's wifi core overheating and it can be burn (I think I'm not professional)

it doesnt overheat that much the max value is set with byte instead of int (idk why)

risinek, 2025-03-19:
it's not about overheating. it was just enough for PoC. You can change the data type from single byte to something larger and adjust parsing on webserver part accordingly