risinek/esp32-wifi-penetration-tool #138: ESP ERROR CHECK

apostolovm, 2025-06-06:
I have successfully  build and flash it on esp32 s3 dev kit, it starts but at some point it gives

ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) .c" line 80
On each attack is a little bit different but it give ESP_ERROR_CHECK on all of them.

Here is a part of the log

`I (52809) sniffer: Starting promiscuous mode...
W (52809) wifi:STA is scanning or connecti
I (52819) wifi:station: 70:a6:cc:7c:56:be leave, AID = 1, reas (52839) wifi:<ba-del>idx:2, tid:0
I (52839) wifi:<ba-del>idx:3zer: Frame analysis started...
I (55869) wifi:<ba-add>idx:2 (ifx:1, 70:a6:cc:7c:56:be), tid:0, TA
hccapx_serializer: Different STA
I (64599) main:attack_handshakhccapx_serializer: Different STA
I (64699) main:attack_handshakhccapx_serializer: Different STA
I (71459) main:attack_handshake: Got EAPoL-Key frame
E (71459) hccapx_serializer: Different S69) hccapx_serializer: Different STA
I (74529) main:attack_handshake: Got EAPoL-Key frame
E (74529) e: Got EAPoL-Key frame
E (74529) hccapx_serializer: Different SI (83969) main:attack_handshake: Got EAPoL-Key frame
I (84049) main:attack_handshake: Got EAPoL-Key frame
I (84069) main:attack_handshake: Got EAPoL-Key frame
I (88989) main:attack_handshake: Got EAPoL-Key frame
I (93579) main:attack_handshake: Got EAPoL-Key frame
I (93759) main:attack_handshake: Got EAPoL-Key frame
E (93759) 69) hccapx_serializer: Different STA
I (93779) main:attack_handnt STA
I (93779) main:attack_handshake: Got EAPoL-Key frame
E I (94089) main:attack_handshake: Got EAPoL-Key frame
I (103509) main:attack_handshake: Got EAPoL-Key frame
I (103939) main:attack_handshake: Got EAPoL-Key frame
I (104019) main:attack_handshake: Got EAPoL-Key frame
I (104039) main:attack_handshake: Got EAPoL-Key frame
I (108529) main:attack_handshake: Got EAPoL-Key frame
I (109239) main:attack_handshake: Got EAPoL-Key frame
) hccapx_serializer: Different STA
I (111539) main:attack_handshake: Got EAPoL-Key frame
E (111539) hccapx_serializer: Different STA
I (111599) main:attack_hands: Stopping promiscuous mode...
I (112809) wifi:ic_disable_sniffer
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) .c" line 80
func: frame_analyzer_capture_stop`

after that it simply restars.

apostolovm, 2025-06-08:
I was able to work around it by commenting the call for error check . When I have some time I'll take a look what is the issue with this call.

bhscodz, 2025-06-08:
i also asked chat gpt , and it said its trying to unregister the errors which were never registered in 2 files esp32-wifi-penetration-tool\main\attack_handshake.c and  esp32-wifi-penetration-tool\components\frame_analyzer\frame_analyzer.c changing them to the actual registered handlers worked for me ```c
void frame_analyzer_capture_stop(){
    //ESP_ERROR_CHECK(esp_event_handler_unregister(ESP_EVENT_ANY_BASE, ESP_EVENT_ANY_ID, &data_frame_handler));
    esp_err_t err = esp_event_handler_unregister(SNIFFER_EVENTS, SNIFFER_EVENT_CAPTURED_DATA, &data_frame_handler);
    if (err == ESP_ERR_NOT_FOUND) {
        ESP_LOGW(TAG, "Handler not found or already unregistered.");
    } else {
        ESP_ERROR_CHECK(err); // Only crash if it's some other error
    }
}``` for  frame analyser.c  and 
``` c ESP_ERROR_CHECK(esp_event_handler_unregister(FRAME_ANALYZER_EVENTS, DATA_FRAME_EVENT_EAPOLKEY_FRAME, &eapolkey_frame_handler)); ``` for attack_handshake.c